pcdshub/pcdswidgets #19: widget sufixes don't match the ones created by PLC vacuum lib

domitto, 2019-08-10:
Hello,
It looks like the PV suffixes between plc lib and the widgets do not match.
I'm attaching the generated db file (from pytmc) and the connection list from the pydm screen.

[gmd.zip](https://github.com/pcdshub/pcdswidgets/files/3489382/gmd.zip)

Thanks,
Diego

hhslepicka, 2019-08-12:
Thank you for reporting the issue, @domitto .
We are updating the suffixes on the widgets and a Pull Request will be out soon fixing the issue.

closed 2019-08-20 by hhslepicka

domitto, 2019-08-21:
There's still a bunch of disconnected pvs, e.g.: :STATE_RBV does not exist in the IOC.
PYDM ca list:

ca://fb:EM1K0:GMD:VRC:10:OPN_SW
ca://fb:EM1K0:GMD:VRC:10:OPN_DI_RBV
ca://fb:EM1K0:GMD:VRC:10:CLS_DI_RBV
ca://fb:EM1K0:GMD:VRC:10:STATE_RBV
ca://fb:EM1K0:GMD:VRC:10:OPN_OK_RBV
ca://fb:EM1K0:GMD:GPI:10:PRESS_RBV
ca://fb:EM1K0:GMD:GPI:10:PRESS_OK_RBV
ca://fb:EM1K0:GMD:GCC:10:HV_SW
ca://fb:EM1K0:GMD:GCC:10:PRESS_RBV
ca://fb:EM1K0:GMD:GCC:10:STATE_RBV
ca://fb:EM1K0:GMD:GCC:10:ILK_OK_RBV
ca://fb:EM1K0:GMD:PTM:10:RUN_SW
ca://fb:EM1K0:GMD:PTM:10:STATE_RBV
ca://fb:EM1K0:GMD:PTM:10:FAULT_RBV
ca://fb:EM1K0:GMD:PTM:10:ILK_OK_RBV
ca://fb:EM1K0:GMD:VRC:20:OPN_SW
ca://fb:EM1K0:GMD:VRC:20:OPN_DI_RBV
ca://fb:EM1K0:GMD:VRC:20:CLS_DI_RBV
ca://fb:EM1K0:GMD:VRC:20:STATE_RBV
ca://fb:EM1K0:GMD:VRC:20:OPN_OK_RBV
ca://fb:EM1K0:GMD:GCC:20:HV_SW
ca://fb:EM1K0:GMD:GCC:20:PRESS_RBV
ca://fb:EM1K0:GMD:GCC:20:STATE_RBV
ca://fb:EM1K0:GMD:GCC:20:ILK_OK_RBV
ca://fb:EM1K0:GMD:PTM:20:RUN_SW
ca://fb:EM1K0:GMD:PTM:20:STATE_RBV
ca://fb:EM1K0:GMD:PTM:20:FAULT_RBV
ca://fb:EM1K0:GMD:PTM:20:ILK_OK_RBV
ca://fb:EM1K0:GMD:VRC:30:OPN_SW
ca://fb:EM1K0:GMD:VRC:30:OPN_DI_RBV
ca://fb:EM1K0:GMD:VRC:30:CLS_DI_RBV
ca://fb:EM1K0:GMD:VRC:30:STATE_RBV
ca://fb:EM1K0:GMD:VRC:30:OPN_OK_RBV
ca://fb:EM1K0:GMD:GCC:30:HV_SW
ca://fb:EM1K0:GMD:GCC:30:PRESS_RBV
ca://fb:EM1K0:GMD:GCC:30:STATE_RBV
ca://fb:EM1K0:GMD:GCC:30:ILK_OK_RBV
ca://fb:EM1K0:GMD:PTM:30:RUN_SW
ca://fb:EM1K0:GMD:PTM:30:STATE_RBV
ca://fb:EM1K0:GMD:PTM:30:FAULT_RBV
ca://fb:EM1K0:GMD:PTM:30:ILK_OK_RBV
ca://fb:EM1K0:GMD:GPI:40:PRESS_RBV
ca://fb:EM1K0:GMD:GPI:40:PRESS_OK_RBV
ca://fb:EM1K0:GMD:GCC:40:HV_SW
ca://fb:EM1K0:GMD:GCC:40:PRESS_RBV
ca://fb:EM1K0:GMD:GCC:40:STATE_RBV
ca://fb:EM1K0:GMD:GCC:40:ILK_OK_RBV
ca://fb:EM1K0:GMD:PTM:40:RUN_SW
ca://fb:EM1K0:GMD:PTM:40:STATE_RBV
ca://fb:EM1K0:GMD:PTM:40:FAULT_RBV
ca://fb:EM1K0:GMD:PTM:40:ILK_OK_RBV
ca://fb:EM1K0:GMD:PRT:40:RUN_SW
ca://fb:EM1K0:GMD:PRT:40:STATE_RBV
ca://fb:EM1K0:GMD:PRT:40:ERROR_RBV
ca://fb:EM1K0:GMD:PRT:40:ILK_OK_RBV
ca://fb:EM1K0:GMD:VRC:50:OPN_SW
ca://fb:EM1K0:GMD:VRC:50:OPN_DI_RBV
ca://fb:EM1K0:GMD:VRC:50:CLS_DI_RBV
ca://fb:EM1K0:GMD:VRC:50:STATE_RBV
ca://fb:EM1K0:GMD:VRC:50:OPN_OK_RBV
ca://fb:EM1K0:GMD:GCC:50:HV_SW
ca://fb:EM1K0:GMD:GCC:50:PRESS_RBV
ca://fb:EM1K0:GMD:GCC:50:STATE_RBV
ca://fb:EM1K0:GMD:GCC:50:ILK_OK_RBV
ca://fb:EM1K0:GMD:PTM:50:RUN_SW
ca://fb:EM1K0:GMD:PTM:50:STATE_RBV
ca://fb:EM1K0:GMD:PTM:50:FAULT_RBV
[... 354 lines cut ...]
fb:EM1K0:GMD:GCC:10:SP_HYS_RBV
fb:EM1K0:GMD:GCC:20:PRESS_RBV
fb:EM1K0:GMD:GCC:20:VAC_SP_RBV
fb:EM1K0:GMD:GCC:20:PRO_SP_RBV
fb:EM1K0:GMD:GCC:20:SP_HYS_RBV
fb:EM1K0:GMD:GCC:30:PRESS_RBV
fb:EM1K0:GMD:GCC:30:VAC_SP_RBV
fb:EM1K0:GMD:GCC:30:PRO_SP_RBV
fb:EM1K0:GMD:GCC:30:SP_HYS_RBV
fb:EM1K0:GMD:GCC:40:PRESS_RBV
fb:EM1K0:GMD:GCC:40:VAC_SP_RBV
fb:EM1K0:GMD:GCC:40:PRO_SP_RBV
fb:EM1K0:GMD:GCC:40:SP_HYS_RBV
fb:EM1K0:GMD:GCC:50:PRESS_RBV
fb:EM1K0:GMD:GCC:50:VAC_SP_RBV
fb:EM1K0:GMD:GCC:50:PRO_SP_RBV
fb:EM1K0:GMD:GCC:50:SP_HYS_RBV
fb:EM1K0:GMD:GCC:60:PRESS_RBV
fb:EM1K0:GMD:GCC:60:VAC_SP_RBV
fb:EM1K0:GMD:GCC:60:PRO_SP_RBV
fb:EM1K0:GMD:GCC:60:SP_HYS_RBV
fb:EM1K0:GMD:GCC:70:PRESS_RBV
fb:EM1K0:GMD:GCC:70:VAC_SP_RBV
fb:EM1K0:GMD:GCC:70:PRO_SP_RBV
fb:EM1K0:GMD:GCC:70:SP_HYS_RBV

PR #20 (2019-08-20): FIX: Readback PVs need to follow PyTMC conventions
Closes #19