pcdshub/pcdswidgets #22: Ebara EV_A pump

domitto, 2019-08-20:
Follow up on a pending email to Nolan:
I'm working on the GMD HMI and I can't find a corresponding widget for the PRT:40 ebara EV_A pump.
The only options for pumps are IonPump, TurboPump, ScrollPump or GetterPump.
Is there any other pump with the same suffix or that requires a new roughing pump widget? 

Thanks,
Diego

hhslepicka, 2019-08-20:
Hi @domitto , please use the ScrollPump for that.

domitto, 2019-08-21:
That doesn't seem to match my IOC, these are the pvs associated with PRT40:
ioc-tst-ads> dbgrep *PRT*
fb:EM1K0:GMD:PRT:40:REMOTE
fb:EM1K0:GMD:PRT:40:REMOTE_RBV
fb:EM1K0:GMD:PRT:40:ALARM_OK_RBV
fb:EM1K0:GMD:PRT:40:RUN_DI_RBV

This is what the scroll pump widget is looking for:
ca://fb:EM1K0:GMD:PRT:40:RUN_SW
ca://fb:EM1K0:GMD:PRT:40:STATE_RBV
ca://fb:EM1K0:GMD:PRT:40:ERROR_RBV
ca://fb:EM1K0:GMD:PRT:40:ILK_OK_RBV

hhslepicka, 2019-08-21:
@ghalym do you mind to check the information? Yesterday you told me that ScrollPump sufixes would match for the Ebara Pump.

hhslepicka, 2019-08-21:
This is a https://github.com/slaclab/pytmc issue.

klauer, 2019-08-22:
`STATE_RBV` is still not in the vacuum library, but everything else for the PRT40 (Ebara Pump) should match up now as the issue was fixed in pytmc. Should we wait to close this until the state one is resolved?

Thanks for reporting, and sorry you have to keep running into these things, @domitto!

hhslepicka, 2019-11-13:
Closing issue as the equivalent widget was pointed out.

closed 2019-11-13 by hhslepicka