pcdshub/pcdswidgets #26: Warnings when Opening Designer

ZLLentz, 2020-01-14:
## Expected Behavior

Things should ideally initialize without warnings. Sometimes they are real issues, sometimes they are not.

## Current Behavior

When opening `designer` from the latest `pcds-3.0.0` environment:
```
QWidget::setLayout: Attempting to set QLayout "" on QFrame "interlock", which already has a layout
Could not display value 'None' using displayFormat 'Exponential' at widget named 'pressure'.
QWidget::setLayout: Attempting to set QLayout "" on QFrame "interlock", which already has a layout
Could not display value 'None' using displayFormat 'Exponential' at widget named 'pressure'.
Could not display value 'None' using displayFormat 'Exponential' at widget named 'pressure'.
Could not display value 'None' using displayFormat 'Exponential' at widget named 'pressure'.
QWidget::setLayout: Attempting to set QLayout "" on QFrame "interlock", which already has a layout
```

## Steps to Reproduce (for bugs)

1. `source /reg/g/pcds/pyps/conda/pcds_conda`
2. `designer`

hhslepicka, 2020-01-14:
That is due to a bug on PyDM that was fixed but a new version was not yet released.
https://github.com/slaclab/pydm/commit/9f5a7c02fd08352aa6728731903fb5879089e41b

ZLLentz, 2020-01-14:
Understood, thanks Hugo. I assumed it was here because everything says interlock/pressure/etc.

closed 2020-01-14 by ZLLentz