pcdshub/pcdswidgets #27: First pass issues for vacuum widgets

slacAWallace, 2020-02-21:
Made a first pass through the vacuum system screen with May Ling this morning and found some issues.

source /reg/g/pcds/pyps/conda/dev_conda
/reg/g/pcds/epics-dev/sabbah01/screens/pydmstuff/HXRVacuum

Need the following addressed for (vacuum) widgets
- PV prefix in tooltip
- PV does not appear in middle click on icons (this is essential especially for valves)
- Icon/ widget states may be inconsistent on ion pumps (waiting for confirmation from @slacAdpai) 
- Adjustable precision for readback fields (need at least one more point of precision and there are inconsistencies in the precision of readback within the pirani cold cathode combo widget)
- Expert screen links

hhslepicka, 2020-02-21:
The middle click just work for controls (buttons, readbacks, etc) not for icons. The middle click is a PyDM behavior, the icons are specifically for pcdsdevices. I will reproduce that feature at the icons for `pcdswidgets`.

hhslepicka, 2020-02-21:
TODO List:
- [ ] PV prefix in general widget tooltip
- [ ] Middle click on icons -> copy PV prefix and show tooltip (same as PyDM widgets)
- [ ] Validate @slacAdpai screen to troubleshoot inconsistent behavior on ion pumps
- [ ] Add option to adjust precision for readback fields
- [ ] Expert screen links

closed 2020-02-22 by hhslepicka