pcdshub/pcdswidgets #30: Expert Screens:  Middle Click Channel Access PVs and Updating Pressure Readings.

slacAdpai, 2020-02-24:
## Expected Behavior

## Current Behavior

## Possible Solution

## Steps to Reproduce (for bugs)

1.
2.
3.

## Context

[image]

## Your Environment

hhslepicka, 2020-02-24:
Thanks @slacAdpai For the issue.
@klauer and I identified this issue late Friday and we are thinking about possible solutions. Typhos has no metadata to select Exponential format and that is why you see the 0. It will be fixed soon.

About the middle click, do you mean when you middle click at a value at the expert screen?

slacAdpai, 2020-02-24:
Hello Hugo,

Thanks for the update about readback displays.

For the middle click, it is for the right column on the expert screens.

Cheers,
Allen

hhslepicka, 2020-02-24:
@slacAdpai about the middle click, I just tested it here using the dev environment and all seems fine.
Would you mind to stop by and we can check it together?

hhslepicka, 2020-02-24:
Job Done! It is now available via `dev_conda`.
The middle click was tested and it is working properly.
[image]

closed 2020-02-24 by hhslepicka