pcdshub/pcdswidgets #36: Depending on location of launching pydm screen the PV precision changes for vacuum devices

slacAdpai, 2020-02-27:
## Expected Behavior
The Precision of the PVs are set to 2 sig figs and should be displayed that way.

## Current Behavior
Currently, the vacuum screen from lfe-console reports exponential values with two significant figures. In xpp-control screen the exponential value is not two sig figs except for the pirani gauges. 
Expert screens also have the same issues, except for pirani gauges.

## Possible Solutions
Implement the two sig fig configuration from the pirani gauge settings to the ion pump, cold cathodes. 

## Steps to Reproduce (for bugs)
1. >ssh lfe-console/xpp-control
2.>source dev_conda
3.>lfe
4. click on hxr vacuum button to launch the HXR vacuum screen.

## Context
[image]

hhslepicka, 2020-02-27:
@slacAdpai This is familiar to me. 
Let me confirm something:
- if it fails when the PVs are consumed through a gateway then the issue is this one: https://github.com/epics-extensions/ca-gateway/issues/11

If not it is something else that will require more investigation

klauer, 2020-02-27:
Oh wow, we really need a gateway update...

hhslepicka, 2020-03-05:
Bruce released a new gateway. I tested the LFE screen via TST network gateway and all the metadata that was not available before is now.
Closing this issue. Please report back if you see it again.

closed 2020-03-05 by hhslepicka