pcdshub/pcdswidgets #48: Fast Shutter widget changes

ghalym, 2020-06-30:
## Expected Behavior
In order to be consistent with the pneumatic valves it should be linked to STATE_RBV  _error_suffix = ":STATE_RBV"  .

## Current Behavior
The fast shutter  _error_suffix = ":ERROR_RBV".

## Possible Solution
 _error_suffix = ":STATE_RBV"

hhslepicka, 2020-06-30:
@ghalym is this just for the fast valve?
Also, how should the two commands work?
Are they both write 1 to open and close?

Could the OPN_SW be transformed in a pv in which 0 closes it and 1 opens and become an ENUM like for others?

ghalym, 2020-06-30:
@hhslepicka yes, only for the fast shutter.
The fast shutter is a double acting solenoid, It reads OPN_SW to actuate the valve open, then CLS_SW to actuate the valve to close. It is different from the other valves.
Can we have two separate command buttons, linked to two command pvs?

hhslepicka, 2020-06-30:
@ghalym no problem. I will get going on it.

PR #51 (2020-06-30): ENH: New Mixin MultipleButtonControl and FastShutter buttons

## Description
- Add a new mixing MultipleButtonControl
- Change FastShutter to use this new mixin.

## Motivation and Context
Closes #48 

## How Has This Been Tested?
Locally

## Where Has This Been Documented?
New documentation entry added.

## Screenshots (if appropriate):
[image]
closed 2020-06-30 by hhslepicka