pcdshub/pcdswidgets #50: After the update to VFS in pcdsdevices and also to FastShutter ind pcdswidgets, Objects not loading properly

slacAdpai, 2020-06-30:
Expert screens are not loading properly.

## Expected Behavior

## Current Behavior
`  File "/reg/g/pcds/pyps/apps/dev/pythonpath/pcdswidgets/vacuum/base.py", line 428, in _handle_icon_click
    display = typhos.TyphosDeviceDisplay.from_class(klass, **kwargs)
  File "/reg/g/pcds/pyps/apps/dev/pythonpath/typhos/display.py", line 1121, in from_class
    'device %s', obj)`

## Possible Solution

## Steps to Reproduce (for bugs)

1. Open HXR Beamline Vacuum
2. Click on an Icon. It cannot bring up the expert screen.
3.

## Context

## Your Environment

hhslepicka, 2020-06-30:
That was due to the update that you executed yesterday at night at pcdsdevices. It introduced a new dependency which was not installed yet ("schema"). Zach installed it this morning and it should be good now.
Please test again and report back here.

slacAdpai, 2020-06-30:
Was schema merged into the repo but not pulled to the working local copy for lucid?

hhslepicka, 2020-06-30:
It was a new external dependency of pcdsdevices.

closed 2020-06-30 by hhslepicka