pcdshub/pcdswidgets #52: More VFS (green/red) Fire!!!! Please!

slacAdpai, 2020-07-15:
## Expected Behavior

[image]

## Current Behavior

[image]

## Possible Solution

Rwar! More Fire!!! hehehe!
[image]

## Steps to Reproduce (for bugs)

1.
2.
3.

## Context

## Your Environment

slacAdpai, 2020-07-15:
Hello Hugo, 
Is it possible to have a larger inverted christmas tree/fireball? I currently have the line width to 1 so some color can be seen. If the width is set to two, there are two specks of color left. I could barely see the color.
-Allen

hhslepicka, 2020-07-15:
@slacAdpai donyou mean make the color area bigger or also make the arrow (black lines) bigger?
I can work on that some time tomorrow.

slacAdpai, 2020-07-15:
Can both the color area and the arrow area (not line widths though) get bigger.

hhslepicka, 2020-07-15:
Line Widths usually will come from stylesheet. I will get this one going and contact you over SLACK for input.

PR #53 (2020-07-15): ENH: Make arrow bigger for FastShutter valve icon.

## Description
Increases the arrow at the FastShutter Icon as requested by @slacAdpai 

## Motivation and Context
Closes #52 

## How Has This Been Tested?
Locally

## Screenshots (if appropriate):
Before:
[image]

After:
[image]
closed 2020-07-15 by hhslepicka