pcdshub/pcdswidgets #59: New Widget Pneumatic Valve with Readback Normally Open

ghalym, 2020-11-18:
TMO has pneumatic valves installed that are normally open with limits switch readback, and we need a new widget for them.
## Expected Behavior
The behavior of this widget is similar to the that of a pneumatic valve. The difference is that it is commanded to close instead of commanded to open.
[image]

hhslepicka, 2020-11-18:
@ghalym about the icon, should we reuse an icon that is already there or do we need to develop a custom one for this widget?

jyin999, 2020-11-18:
I prefer to use the same VGC icon.

ghalym, 2020-11-18:
@jyin999 how about we add the letters NO next to t he top rectangle?

jyin999, 2020-11-24:
@ghalym I like it.

jyin999, 2020-12-15:
@hhslepicka any update?

hhslepicka, 2020-12-15:
@jyin999 I am on vacation. I can work on it early January.

jyin999, 2020-12-15:
@hhslepicka Early January is good. Thank you.  Have a great holiday time!

hhslepicka, 2021-01-04:
@jyin999 and @ghalym does this look like what you are looking for?
[image]

jyin999, 2021-01-04:
@hhslepicka @ghalym Good for me.

PR #60 (2021-01-05): ENH: Normally Open Pneumatic Valve Widget

## Description
Normally Open Pneumatic Valve Widget

## Motivation and Context
Closes #59 

## How Has This Been Tested?
Locally

## Where Has This Been Documented?
Documentation was included in this Pull Request

## Screenshots (if appropriate):
[image]
closed 2021-01-05 by ZLLentz