pcdshub/pcdswidgets #79: Can't install from pypi

ZLLentz, 2023-02-22:
## Expected Behavior

pip install pcdswidgets should work

## Current Behavior

requirements.txt is read during setup.py but is not included in the distribution
```
Collecting pcdswidgets
  Downloading pcdswidgets-0.7.1.tar.gz (46 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.7/46.7 kB 3.1 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [7 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-qzzep_p7/pcdswidgets_0ff2413a26f349cf87dc0822c88e4b64/setup.py", line 5, in <module>
          with open('requirements.txt') as f:
               ^^^^^^^^^^^^^^^^^^^^^^^^
      FileNotFoundError: [Errno 2] No such file or directory: 'requirements.txt'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed
```

## Possible Solution

Do the full repo update we've been doing elsewhere incl switching away from setup.py

## Context
Experimenting with creating py3.11 environments with designer plugin support

klauer, 2023-02-23:
I don't think I've been adding `requirements.txt` (or `dev-` / `docs-`) to `MANIFEST.in` in the recent migration. Either we already have it, or it's unnecessary with `pyproject.toml`... Need to double-check those repositories.

ZLLentz, 2023-02-23:
Right, if the requirements are embedded in the toml file those repos are probably fine, otherwise they probably have this same issue

PR #80 (2023-02-23): DEV/MNT: migrate to GitHub Actions and pyproject.toml [LCLSPC-603]

## Description
* Upgrading by way of https://github.com/pcdshub/pcds-python-migration-tools
* pcdswidgets no longer uses Travis CI and has migrated to GitHub Actions for continuous integration, testing, and documentation deployment.
* pcdswidgets has been migrated to use setuptools-scm, replacing versioneer, as its version-string management tool of choice.
* pcdswidgets has been migrated to use the modern ``pyproject.toml``, replacing ``setup.py`` and related files.
* Older language features and syntax found in the repository  [... cut]
closed 2023-02-23 by klauer